simon-jouet/ESP32Controller #1: Modifications for the R2

simon-jouet, 2018-07-08:
Somes notes for the R2 to consider

* Keep the ADS1015? Either bad batch in the first run or very temperature sensitive when soldering...
* Add a reset button instead of relying only on DTR
* More compact PCB
* Keep 24V automotive fuses? If so change the socket to something easier to solder. Maybe move to micro autotomotive fuses with sockets 
* Increase the resistance of the witness LEDs as they are super bright at the moment
* Move the SD card slot to the top so the pins can be waved soldered

keyz182, 2018-07-08:
> Add a reset button instead of relying only on DTR

I'd suggest not only a reset button, but two pads or holes for a user to solder a 2 pin header to for wiring up a reset button. Reset is often used as an emergency stop, as it'll stop everything, but bring the software back up to ensure fans are kicked in where appropriate (e.g. hotend, to prevent heatcreep, and subsequent jams).

> Increase the resistance of the witness LEDs as they are super bright at the moment

Definitely! Crazy unnecessarily bright LEDs are getting out of hand :p

As the board sits currently, is it possible to pull a PWM 5v signal from somewhere (e.g. heater)? I ask because it would be interesting to use this board to run a laser cutter, which generally require a 5v PWM for intensity. For e.g. the Duet, this is done by disabling the/a heater, and using it's PWM signal.

simon-jouet, 2018-07-08:
> I'd suggest not only a reset button, but two pads or holes for a user to solder a 2 pin header to for wiring up a reset button. Reset is often used as an emergency stop, as it'll stop everything, [...]

So that part is actually already the case, I broke out the pins of the ESP32 so the two top left pins in the header can already be used for that :). At the moment I reboot the board by shorthing both pins with my tweezers...

> As the board sits currently, is it possible to pull a PWM 5v signal from somewhere (e.g. heater)? I ask because it would be interesting to use this board to run a laser cutter, which generally [...]

At the moment the board runs at 3.3V however there is 5V but it's not exposed (from USB or the SMPS) and regulated down to 3.3V. How much current do you need for the PWM, is it just singalling? because every output pin on the ESP32 can be PWM you can just use any of the available free pin (which isn't much) at 3.3V. You could also use the SX1509 GPIO expander that actually support PWM out - kind of depend how often you need to change the pwm frequency but that would also be at 3.3V. The SX1509 could be powered at 5V to get 5V IO but then would need level translation when talking to the ESP32. I would guess most laser that have a PWM signal input would be quite happy with 3.3V instead of 5V as most devices have high a threshold lower than that (but i'm guesssing - i have a laser module in my aliexpress cart just not ordered yet :))

keyz182, 2018-07-08:
> So that part is actually already the case, I broke out the pins of the ESP32 so the two top left pins in the header can already be used for that :). At the moment I reboot the board by shorthing both pins with my tweezers...

Awesome :)

The PWM is just signalling, so low current. Not sure what would happen at 3.3v, but yeah, it might work, would have to test. No biggie though to add an external level shifter on a separate board for that kind of thing. It's what I've just done on my Duet :)

julianobst, 2018-07-09:
According to the chapter "2.2.1.2 Positioning a ESP32 Module on a Base Board" from the [ESP32 Hardware Design Guidelines](https://www.espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf), it would be better for the rf performance to remove the copper under/around the antenna or better place the modules antenna outside the baseboard.

simon-jouet, 2018-07-09:
> According to the chapter "2.2.1.2 Positioning a ESP32 Module on a Base Board" from the ESP32 Hardware Design Guidelines, it would be better for the rf performance to remove the copper under/around the antenna or better place the modules antenna outside the baseboard.

Yeah you're right, the current placement is quite far from optimal. I have removed the ground plane on the top layer but not nearly enough to align with the hardware guidelines. Thanks for that I guess with the next rev the ESP is going to move to the right of the board and the antenna is probably going to stick out (which i'm not a big fan off)

julianobst, 2018-07-09:
I think a cutout under the antenna would also do the job. With a cutout you could keep the alignment of the board outline and antenna outline.

simon-jouet, 2018-07-22:
Hey guys, thinking about the R2 of the board - any suggestions on which fuses to go for? I like the automotive fuses like in the current board but they take a *LOT* of space. Did you ever had to replace a fuse - if so how often? (I never had to on my printers/boards).

I very much like https://uk.rs-online.com/web/p/fuse-blocks/7693716/ but they are not cheap. On aliexpress some 1808 SMD fuses are significantly cheaper but they are without a holder an directly soldered on the PCB.

keyz182, 2018-07-22:
They look cool, but also really fiddly to change. 

The newer version of the Duet Wifi uses blade type fuses, but they're smaller than the normal automotive fuses (see the 3rd screenshot on their store - https://www.duet3d.com/DuetWifi).

simon-jouet, 2018-12-20:
I found tiny 1808 fuse holders on aliexpress for quite cheap, I ordered a bunch and should get them in January, if they are nice I might use this in the next revision of the board.

https://www.aliexpress.com/item/Hzy-Gold-foot-1808-125V-0451-SMD-Fast-blow-Fuse-0-5A-0-75A-1A-2A/32872911147.html?spm=a2g0s.9042311.0.0.76184c4dnU4DP4

simon-jouet, 2019-07-06:
closing #8

closed 2019-07-06 by simon-jouet