simon-jouet/ESP32Controller #3: Request for BOM

vivian-ng, 2018-11-04:
Hi, do you have a BOM? I am trying to figure out what are the specifications of the inductor, mosfets, microSD card shield, diodes, transistors, etc. that you are using in your design.

simon-jouet, 2018-11-05:
Hi @vivian-ng, I've seen some of your messages, sorry for not getting back to you the last few months have been very hectic and I had very little time to dedicate to this. I don't have a proper BOM list yet but I have records of everything i've used.

- Inductor: https://uk.farnell.com/webapp/wcs/stores/servlet/ProductDisplay?catalogId=10001&langId=44&urlRequestType=Base&partNumber=2328706&storeId=10151
- Mosfets: https://www.aliexpress.com/item/New-IRLR8743-TO252-MOSFET/1695115706.html?spm=a2g0s.9042311.0.0.27424c4dlT22Rw
- microSD: https://www.aliexpress.com/item/10PCS-Push-Push-Type-TF-Micro-SD-Card-Socket-Adapter-Automatic-PCB-Connector-100-brand-new/32881919643.html?spm=a2g0s.9042311.0.0.27424c4dlT22Rw
- Diodes: (I think it's in the schematic? IIRC i didn't populate the TVS diodes in the dev version)
- 100uf caps https://www.aliexpress.com/item/E021-50pcs-100UF-35V-6-3mm-7-7mm-SMD-electrolytic-capacitor/32712209859.html?spm=a2g0s.9042311.0.0.27424c4dlT22Rw
- screw terminals: https://www.aliexpress.com/item/10pcs-lot-KF7-62-7-62MM-Pitch-2Pin-PCB-Screw-Terminal-Block-Connector-KF7-62-2P/32863208285.html?spm=a2g0s.9042311.0.0.27424c4dlT22Rw (4 and 6 pins versions)
- transistor for the usb to serial https://www.aliexpress.com/item/100PCS-S9014-MMBT9014-9014-Mark-J6-SOT23-SOT-23-NPN-General-Purpose-Transistors-0-1A-50V/32854935087.html?spm=a2g0s.9042311.0.0.27424c4dbPHaz4

The rest should be just normal 805 or 603 passives so pretty much anything will do

vivian-ng, 2018-11-06:
@simon-jouet Thank you for giving the links! This will really help me in sourcing for the items.

lkvapil, 2018-11-13:
Hi,

could you share your BOM @vivian-ng ?

Thank you
Lukas

simon-jouet, 2018-11-17:
Guys if you are building one of those boards you can now ignore the ADS1015 and C16. I've figured out a way to get good temp measurements directly from the ADC on board of the ESP32 (especially if you use a revision 1).

I wanted to get rid of the ADS1015 anyway, it's not super cheap and it's a pain to solder (at least the ones I got, half of them didn't work) + we need custom support in marlin.

If you just want to get started with the ESP32 you can also ignore the GPIO expander + supporting circuitry for now. I still haven't added the support in marlin for that (but I will - this gpio expander is very nice, just a bit of a PITA to solder).

Would be great if you could get your own boards - I've been printing with mine for about 50 hours (maybe more considering the number of calibration cubes i've printed). I will try to work on marlin in the coming weeks to add support for some stuff ("eeprom", adc improvements, servo?), once the software is more mature and based on the feedbacks let's design a R2

vivian-ng, 2018-11-19:
@simon-jouet Thanks for the update! I am in the process of putting together a minimalist ESP32 board based on your design, without the GPIO expanders and such as they are not supported in Marlin. Just 4 motors, 1 hotend, 1 print fan, 1 heatbed. Have you tried @luc-github [ESP3D-adapted branch on his Marlin fork](https://github.com/luc-github/Marlin/tree/eps32_webui)? I think it is quite promising for a bare minimal ESP32, since it allows the ESP32 to be controlled via WiFi, given that it doesn't have enough pins to support an LCD controller. Of course, I am going to break out the TX/RX pins so someone with a MKS TFT can easily use it. But ESP32 running on a ESP3D-enabled Marlin firmware just sounds cool. Meanwhile, while designing the minimalist board, I am waiting for the mosfets to arrive (ordered on Aliexpress) so that I can try a jury-rigged setup.

For EEPROM support... that was one of the things that I realized too when I compiled @luc-github fork. This [article](http://tinkerman.cat/eeprom-rotation-for-esp8266-and-esp32/) has a link to an Arduino library for emulating EEPROM on ESP32.

luc-github, 2018-11-19:
I think EEPROM no need to be added there is a space on flash of 4KB for such purpose and as now ESP3D for Marlin use nvs partition instead of EEPROM partition to store settings, this space is fully free

vivian-ng, 2018-11-20:
@luc-github Yes that is basically why I pointed out that article. Arduino Core already has libraries to use ESP32's flash memory as "EEPROM" and this article has an improved version of such a library which I think can be considered for use by any who wants to add "EEPROM" support for ESP32. I was scanning through Marlin and I think the EEPROM-related code is all in one file inside the HAL subdirectory of the respective platforms.

simon-jouet, 2018-11-20:
For the screen you could use an ssd1306 oled screen I think I tried at some point and it works, I might give it another shot though.  Still it doesn't really solve the extra pins necessary if you want a rotary encoder or you might just make it with no pins to spare. My current setup is very similar to yours just a prusa i3 clone with 5steppers (2 for z in parallel) and the usual hotend, part cooling fan and heatbed with an inductive probe.Just a word of caution with your setup, it's what I did at the beginning but make sure you have a clean ground between all the components and try to plug everything (USB host + psu) on the same socket or you might have ground loops. 

For the EEPROM emulation, called PersistentStore in marlin, I'm not sure yet if I want to use nvs or spliffs. The marlin api is much more similar to a file seek+write than the blob of nvs and I would rather not use too many write cycles of the flash, anyway I will have a look at that soon.

lkvapil, 2018-12-02:
Hi @simon-jouet, 

are you still developing new version or it's in the states, that everything is working ?

Thank you

simon-jouet, 2018-12-02:
Hi @lkvapil,

Yeah I will probably make a second revision of the board at some point, but at the moment it does enough to port Marlin and things might change depending on what can and can't be done in software, so I'm not rushing for a new hardware revision and instead spend some time on the software (even though time has been very limited recently)

At the moment the board and Marlin works well enough to print (it's has been the fulltime board on one of my printers for about 6 months) but there are a lot of features that needs to be implemented and tested. I was hoping to have some people helping with the Marlin HAL but it seems that a lot of people have been focusing on STM32 boards instead (not too sure why, I think the ESP32 is superior in many ways although the number of IOs is a bit of an issue). Anyway, if you want to test the ESP32 for printing either get the current board and it will work or do a makeshift one like @vivian-ng and that should work too.

I'm also open to feedback on the second revision of the board (there is another issue about that). I really like the big 7.68mm connectors I used for power and mosfets because they deal with the high current without issues and you can use crimps which makes everything much nicer. Not sure everybody likes it though. I want to change the fuses to something smaller either micro automotive fuses or go for the 1208 SMD fuses (but then it's hard to swap)... One of the "main" block for a second revision right now is to get the GPIO expander to play ball with Marlin which might change some of the GPIOs allocation and therefore would need another revision of the board. I don't think it's that bad to do, just didn't really had time to work on that

vivian-ng, 2018-12-09:
@simon-jouet Just an update. I put together a Morpheus board and used that to test out the Marlin branch which @luc-github is working on. One thing I realized was the Morpheus board did not have a pull-up resistor for MOTOR_ENABLE, which caused the motors to spin when the ESP32 starts up.

Anyway, I am designing a basic ESP32 printer board (no GPIO extender, etc.) based on the Morpheus design, this ESP32Controller, and RAMPS (plus a few other 32-bit boards with schematics found online). Once the design has been confirmed to be working, I will share it on Github.

vivian-ng, 2018-12-10:
@simon-jouet @luc-github The [schematic for my ESP32-based board](https://github.com/vivian-ng/Hardware/blob/master/MRR_ESP_3DP/Schematic_MRR_ESP_3DP_v0.3.pdf) is now on Github. This is the first version which I feel comfortable with sharing, looking forward to feedback.

v0.1 was a total failure. Don't ask.
v0.2 didn't have the pull-up on MOTOR_ENABLE, and uses the 7.62mm pitch terminals given by @simon-jouet above. But I find them taking up too much space, so I went with 5.08mm pitch terminals like those found on MKS boards (which I own, and have no issues so far with the terminals) for v0.3.

v0.3:
- Basic, just ESP32 with a CH340 so that it can be accessed via USB.
- 5.08mm pitch terminals
- Pull-up on MOTOR_ENABLE
- AUX1 connector, in case anyone wants to connect to a MKS TFT, since there simply aren't enough pins for a Reprap LCD or any other type of controller using a rotary encoder.
- Design consideration is to fit this within 100mm by 100mm (currently, I am working on the PCB layout) so that people can take advantage of cheap PCB production (there is a site which offers 10 PCBs at $2 if the size is within 100mm by 100mm).
- TODO: rename all "+12V" to "VIN" and "+24V" to "VBED" which to me is largely cosmetic, but probably an important detail to real PCB designers.

Any feedback is welcome so that I can make the necessary changes before sending this for a test fabrication.

misan, 2018-12-10:
Having 24V available, why don't you use it too for powering the motor
drivers? You would get more torque at faster speeds, which is good.

vivian-ng, 2018-12-10:
@misan 
> Having 24V available, why don't you use it too for powering the motor
drivers? You would get more torque at faster speeds, which is good.

Which is why I think a professional PCB designer will likely say my naming conventional is very poorly done. I used "+12V" and "+24V" to differentiate between two inputs, both of which can be 12V to 24V.

The way to read "+12V" is "12V to 24V".
The way to read "+24V" is also "12V to 24V", just connected to another part of the circuit.

I really need to get around to renaming it.

That said, I am also a bit uncertain about how much power the stepper drivers and motors will draw. The terminal for the heat bed is going to be passing a lot of current, and one of the reasons why I didn't connect VMOT to the "+24V" rail is because I don't want too much current through that terminal.

misan, 2018-12-10:
Consider each motor can draw around 1A.

vivian-ng, 2018-12-12:
I have changed the inputs to VIN and VBED instead of +12V and +24V, plus added a prelim PCB design. Uploaded [here](https://github.com/vivian-ng/Hardware/tree/master/MRR_ESP_3DP). All feedback welcome. Will be sending this for fabrication soon, so that I can give it a test.

Also, I created a [thread](https://github.com/vivian-ng/Hardware/issues/1) on the repo itself for further discussion on my board, so that I don't hijack this repo, which is for Simon's board.

simon-jouet, 2019-07-06:
closing this, the BOM is now in #8 and you have the thread on your repo

closed 2019-07-06 by simon-jouet