simon-jouet/ESP32Controller #7: R2 routed?

KanyonKris, 2019-06-28:
@simon-jouet, very nice project!
The schematic looks nice. I opened the PCB layout and I see the part footprints but no traces, just the net lines. Is this correct or do I have something wrong?

simon-jouet, 2019-06-28:
Hi @KanyonKris

Thanks, I haven't routed the R2 board because I'm still tweaking it a bit, if you want to see a routed version @Ringel pushed a MR for one. I haven't tried it but if you wait a bit @Ringel should be able to tell you if it works or not.

I've done few more Tweaks to the schematic since then and will continue a bit over the weekend. I also think I've found a decent layout so the next step will be to clean that up a bit and start the routing. With the layout I've drafted for now we should have the full board in the same footprint as an arduino mega

KanyonKris, 2019-06-28:
Thanks for the reply. I'll check out Ringel's board and watch for your updates.

simon-jouet, 2019-07-06:
closing #8

closed 2019-07-06 by simon-jouet