simon-jouet/ESP32Controller #8: R2 preview and BOM

simon-jouet, 2019-07-06:
Okay so I've finished the R2 and ordered some PCBs, here is the 3d render

[image]

I started the BOM and my current estimated cost for the components + PCB is about ~£10 ($12.5) of course on top of that there is the assembly cost and wastage but that's very dependent on the number of boards being made (and for the time being i do it myself)

[BOM.xlsx](https://github.com/simon-jouet/ESP32Controller/files/3364564/BOM.xlsx)

While doing the BOM i've noticed I've made two mistakes in footprints for the diodes, I was hoping to get a 603 package for the diodes but that's not the case, shouldn't be too much a problem for this batch but I will have to get it fixed.

shawe, 2019-07-20:
Thanks for sharing your work.
Do you try it (or can do it) with this GRBL fork? https://github.com/bdring/Grbl_Esp32 Looks interesting have a same board that can work well with both projects.

Aggebitter, 2019-07-21:
Is there a possibility to by one assembled or un assembled from you?

Aggebitter, 2019-07-21:
@shawe The GRBL_ESP32 does not use shiftregisters for the stepper motors. But you can configure Marlin to work as an router/Mill

shawe, 2019-07-21:
Thanks for your answer @Aggebitter, then Marlin is more interesting to me at least for start. I asume that each project are more specific to each work, but don't look for if someone can do both.

I'm interested also in get one, I try a quote from jlcpcb and seems to be available 5 boards at 2$ (offer only for the first buy and also applicable for more units) but I'm not familiar to solder SMD components and also don't have the needed tools to work with that precision.

Aggebitter, 2019-07-26:
@shawe I'm running GRBL_ESP32 on my small router, even tryed to get GRBL_ESP32 interested in using shiftregisters but no response on there part.

vivian-ng, 2019-07-31:
@simon-jouet Can you share the settings you had to make in Configuration.h and pins_ESP32.h to [test out the OLED](https://github.com/MarlinFirmware/Marlin/issues/14345#issuecomment-515755280) and possibly ILI9341? I am wondering about the display/LCD option to set in Configuration.h, and the settings for SDA pin, SCL pin, I2C address for the OLED display, and the different pins for the ILI9341.

simon-jouet, 2019-07-31:
> Is there a possibility to by one assembled or un assembled from you?

For the time being I don't have enough components to assemble more than a handful of boards so probably not (I will send one to thinkyhead and probably to Luc for dev). I've ordered a few more components so i might be able to assemble a few more boards. Let's see when I get the components but I would rather not assemble too many myself as it takes a while (maybe ~2 hours per board) and sending the individual components not soldered will be a mess with the amount of SMD resistors/cap/ics. If enough people are interested we could potentially get a batch assembled in China but I'm not too sure what would be the price of that.

> @simon-jouet Can you share the settings you had to make in Configuration.h and pins_ESP32.h to test out the OLED and possibly ILI9341? I am wondering about the display/LCD option to set in [...]

For the OLED it's relatively straightforward, just connect the screen D0 and D1 to SCL and SDA in the esp32 (pin 21, 22), connect the enable to the enable pin of the ESP32 and tie D/C to ground (https://electronics.stackexchange.com/questions/164680/ssd1306-display-spi-connection-or-i2c-according-to-resistors). Once that's connected in your configuration.h comment out `//#define U8GLIB_SSD1306` and modify your platformio.ini for the esp32 like that

```diff
 lib_deps =
+  ${common.lib_deps}
   https://github.com/me-no-dev/AsyncTCP.git
   https://github.com/me-no-dev/ESPAsyncWebServer.git
 lib_ignore  =
-  LiquidCrystal_I2C
   LiquidCrystal
-  NewliquidCrystal
-  LiquidTWI2
   TMC26XStepper
   c1921b4
-  SailfishLCD
-  SailfishRGB_LED
```

For the ILI9341 I haven't really tried yet, I had a quick look but the current code is quite specific for STM32 IIRC so there will be some dev required for that (and it's not very high in my priorities to get that working for now)

shawe, 2019-07-31:
> > If enough people are interested we could potentially get a batch assembled in China but I'm not too sure what would be the price of that.

@simon-jouet I'm asking for a quotation on PCBWay of 20 units (to get an idea), comparison price later. Maybe 20 are a lot of they at this time, I don't know, but is the most affordable option for few units.

Also, maybe there are better alternatives to make it, but this company can do it all process without wasting your time. I upload your gerber files, and your BOM list and maybe tomorrow they fill it the quotation.

Quotation at this moment:
- 2 Layers Size 102x71mm 1.6mm Product No.:W234586ASH1 [PCB Production]
  -  20 units at $37.00 - $1.85/unit
  -  5 units at $24.00 - $4.80/unit
- SMT for PCB No. W234586ASH1 [PCB Assembly ]
  -  20 units at $30.00 - $1.5/unit
  -  5 units at $30.00 - $6/unit
- Components
  - Waiting for reply of PCBWay (I edit this when receive a notification)
- Shipment
  - Unknown at this moment
- Your fees for your work

They also promoted open source: Share&Sell _This feature is for open source enthusiasts who would like to share the cool projects to others and earn commission._

Maybe this last option is more comfortable to you, and also allow us to get one fully assembled.

If anyone is interested in how this company work, I found this promoted video that is a tour on this factory https://www.youtube.com/watch?v=24ehoo6RX8w

simon-jouet, 2019-08-02:
Hi @shawe,

Thanks for looking into that, I had a brief look couple of days ago at elecrow that provides similar services but didn't go as far as asking for a quote. 20 might be reasonable, I might post on Twitter in the next few days to check who would be interested and point them to here, even if there are couple extras it's not too big a deal.

I will revise the the BOM looks like there are quite a few things that are to be fixed. But looking at the prices it seems more expensive that what I would have expected, let's see once I revised the BOM and let's see what this includes. Do you know what the total of $452 includes?

Before we go for a batch I need to slightly revise the current PCB there are two footprints that are incorrect in the current board (two diodes) and I want to add a couple of headers for the +5V and +3.3V (realised this when I tried to connect a BLTouch...)

I think the open source approach is good, it will save in admin time plus if I need to ship the board from the UK it's going to be quite expensive.

shawe, 2019-08-02:
I'm not sure with the prices, because they change it without reason and without any explanation, and not respecting what I see first day, but looking their page seems to be this:

- $37 for all PCB Production
- $410 for all PCB Assembly (this part appear first time to $30 for <5, but seems to be per unit)
- $452 for all PCB components, the double than is in your list (this part is not yet filled on the quotation, I suppose you see it on the replyed BOM)

Using this prices as reference, the board fully assembled seems to be $44.95/unit

This appears on my quotation (it's in spanish):
[image]

@simon-jouet if you prefer it, you can upload this files and be directly in contact with PCBWay to clarify all that you want, you only need some minuts to upload your files, and wait for the reply as I attached here.

simon-jouet, 2019-08-02:
Looks very very expensive to me, I might be missing something but here the assemble board is about 3 times the BOM which I think is a bit unreasonable. (the BOM is wrong and a bit too high but still wouldn't change much). My BOM is ~£10 ($12) including the PCB with components that have been shipped individually from China. Here the cost components is higher while they weren't shipped, and the BOM price doesn't include the PCB. Finally we probably need to add shipping + import duties on top of that.

The prices in my BOM are in GBP (£) and they use ($) so their is some difference from that, but even with that there is quite a big difference. 

I might check with Elecrow to see how much that would be and I might ask around to see if people have some info about this and if anyone has some experience with that please let me know :)

shawe, 2019-08-02:
I think the same, it's a big difference, maybe they are using another components in comparison with your list. Maybe later I can check component to component to found the problem comparing both BOM list.

The good part, is that they include part numbers, and can be checked or change it to some specific parts, reducing this difference but require an additional check that maybe is also need with Elecrow.

shawe, 2019-08-02:
I use the BOM replied by PCBWay, and fill it with Mouser and Digikey prices (always per unit, I know is the most expensive way) and probably I do a mistake some parts. Also there are some parts that I don't know how to found it.

Anyway, the final price is a little less than buying all individually, but of course not at same price than buying at the next level for quantity and also DIY.

[W234586ASH1-20sets-BOM 2019-8-1.xlsx](https://github.com/simon-jouet/ESP32Controller/files/3462677/W234586ASH1-20sets-BOM.2019-8-1.xlsx)

The PCB Assembly on Elecrow must be very similar, because they also have an online estimate quotation, and the price is calculated based on unique parts, holes, SMT pads, ... I don't fill this form on PCBWay, I only attach the BOM that is incomplete to a quotation and wait for a more realistic quotation.

If I'm not wrong, this form is not well completed on the other site, because I get the same start price with Elecrow. It's filled like in this image:

[image]

shawe, 2019-08-16:
Sorry for saying this again, but people from PCBWay are asking me again if the BOM list was reviewed with things they attached on file to be able to finish the quotation.

closed 2021-07-06 by simon-jouet