simon-jouet/ESP32Controller #9: Malyan, and monoprice have been paying attentoin and made a control board using the ESP32 SoC

Sniffle, 2019-07-07:
This is the control board in use on the Monoprice MP10 and MP10 Mini, which is rebranded from Malyan with part numbers MA10 and MA10 mini.

https://imgur.com/a/0MB0AIy

If the imgur album link fails, I will make the images public and link to them directly.

The SoC in use seems to be the ESP8266EX, which has a touch display attached to its back, a daughter board for USB and SDCard.  A 16pin ribbon cable runs to the heater and stepper board.  power is input from the power supply to the stepper control board and send along the 16 pin cable to the ESP32 main control.  

That's as far as ive had a chance to dig into this.

If this piques your curiosity I'd be happy to put time towards helping get this done.  If there is any information, pictures, diagnostics, etc.  Id be happy to help.  

display and ESP control board - https://www.monoprice.com/product?p_id=39107
Stepper heater board - https://www.monoprice.com/product?p_id=39106

simon-jouet, 2019-07-10:
Thanks @Sniffle for the info, it would be good to have a better look at it.

My first impression is that the esp32 is not used for printing but is used a the "screen" for the printer. My guess is that the ESP32 drives the screen and provides a webUI but the printing is done by another microcontroller (likely an atmega) and the ESP32's serial is connected to the Atmega's serial.

I might be totally wrong and would be nice to see a manufacturer using the ESP32 directly. But if you look at https://images.monoprice.com/productlargeimages/391062.jpg you can see below the capacitor a QFP package with what looks like 64 pins.

Ringel, 2019-07-10:
Its a ESP8266 - not a ESP32!
Yes looks like it drives the display ... maybe via the 2 8bit shiftregisters 74HC164 and the XPT2046 Touchcontroller 

Something like https://github.com/luc-github/ESP3D might be running on it.

Sniffle, 2019-07-10:
I'll get some better pics of the driver board.

My initial though was also that it was display driving an atmega but I couldnt find the chip initially.  I'll take better pics and see what pops out

closed 2021-07-06 by simon-jouet