simon-jouet/ESP32Controller #13: ESP32 wifi mode problem

reddomon, 2020-01-26:
Hi. When I compile if wifi is enabled, ESP32 is resetting. Try different libraries and different code stack nothing solved the problem. I am sending M20 via serial. Then M23 some.gcode and after sending M24 start gcode command an unknown reset occurs. Reset reason was Heap Corruption.

reddomon, 2020-01-26:
Here is the decoded stack trace:

PC: 0x40084aa4: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 93
EXCVADDR: 0xa5a5a5c1

Decoding stack results
0x40084aa4: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 93
0x40084b31: heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 140
0x40084ff5: _malloc_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 37
0x401526ff: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 50
0x400e16a1: AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .piolibdeps\esp32\ESP Async WebServer\src\AsyncWebSocket.cpp line 1207
0x400e1e40: AsyncWebSocket::textAll(char const*, unsigned int) at .piolibdeps\esp32\ESP Async WebServer\src\AsyncWebSocket.cpp line 960
0x400e1e55: AsyncWebSocket::textAll(unsigned char*, unsigned int) at .piolibdeps\esp32\ESP Async WebServer\src\AsyncWebSocket.cpp line 1096
0x400d260b: WebSocketSerial::write(unsigned char) at Marlin\src\HAL\HAL_ESP32\WebSocketSerial.cpp line 134
0x400d350f: serialprintPGM(char const*) at Marlin\src\core\serial.cpp line 37
0x400d5976: GcodeSuite::host_keepalive() at Marlin\src\gcode\gcode.cpp line 883
0x400d323e: idle() at Marlin\src\Marlin.cpp line 671
0x400dad84: Planner::synchronize() at Marlin\src\module\planner.cpp line 1541
0x400d66d1: GcodeSuite::G4() at Marlin\src\gcode\motion\G4.cpp line 36
0x400d529b: GcodeSuite::process_parsed_command(bool) at Marlin\src\gcode\gcode.cpp line 208
0x400d5901: GcodeSuite::process_next_command() at Marlin\src\gcode\gcode.cpp line 826
0x400d6ef0: GCodeQueue::advance() at Marlin\src\gcode\queue.cpp line 885
0x400d329d: loop() at Marlin\src\Marlin.cpp line 1173
0x400e9f69: loopTask(void*) at C:\users\red\.platformio\packages\framework-arduinoespressif32\cores\esp32\main.cpp line 19
0x40088c65: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

reddomon, 2020-01-26:
Solved disabling websocket extension. Web server works as expected, async web file upload works as expected but couldn't find the reason behind the reset issue.

vivian-ng, 2020-01-27:
@reddomon Just to get some details so that others can provide better responses.

What version of Marlin are you using? 2.0.x or bugfix-2.0.x? If possible, which commit (or around what date of the commit)? Marlin bugfix-2.0.x is under active development, so it helps to test with latest commit of bugfix-2.0.x to see if the problem still exists.

reddomon, 2020-02-23:
Sorry about the delay. I am using Marlin bugfix-2.0.x branch updated recently.

closed 2020-04-25 by reddomon