simon-jouet/ESP32Controller #17: Generic I2S questions

petervanderwalt, 2021-11-11:
Hi!. I am experimenting with TMC5161 drivers, on a breadboard setup running an ESP32 with the I2S expanders.

I have just about everything else working, i can control dumb drivers just fine, but I don't seem to be getting the CS pin to switch for spi transactions to the TMCs.

Verified with a scope, activity on SCK and MOSI, but CS just stays high. Do you know if its possible to use on of the I2S expansion pins as a CS pin for the Trinamics (no spare GPIOs on the ESP32, and I know Esp32_Grbl can use the I2S pins as CS pins for TMC and

M122 reports ALL HIGH so I know its a comms issue, and not seeing CS go low at the start of the transaction makes me suspicious of whether I can use I2S CS pins in Marlin?

Also, can I run more than 32 bits? For example, can I run 6* 595s for 48 extra Outputs? (step/dir/enable/cs for drivers; fets for heaters, laser, fans, rgb leds; rc-servo/bltouch, etc)

Can one run Servo's off the I2S?

simon-jouet, 2021-11-11:
Hi @petervanderwalt 

> Do you know if its possible to use on of the I2S expansion pins as a CS pin for the Trinamics (no spare GPIOs on the ESP32, and I know Esp32_Grbl can use the I2S pins as CS pins for TMC and
> M122 [...]

So I've personally just used GPIO's on the ESP32 for CS as timing wise use the GPIO expander can be finicky but there is no real reason why it wouldn't work (except timing). If the line stays high then there might be something else going on. Did you configure Marlin properly to have the first 16 bits as the stepper stream and the other 16 bits as the "pins"? If you have the default configuration IIRC all 32bits will be the steppers stream which won't allow you to use them as GPIOs.

If you are still in the experimentation process you could try with an ESP32-S2, I've got a version of Marlin for it too (not pushed yet) and it gives access to quite a few more GPIOs which is nice.

> Also, can I run more than 32 bits? For example, can I run 6* 595s for 48 extra Outputs? (step/dir/enable/cs for drivers; fets for heaters, laser, fans, rgb leds; rc-servo/bltouch, etc)

Out of the box no, but making the I2S code work over 64 bits instead of 32 should be fairly straightforward.

> Can one run Servo's off the I2S?

Kind off but I wouldn't recommend it. You would need to have a PWM software implementation for the I2S stream, it would be fairly straightforward to code but it would also use CPU cycles. For some things that are timing sensitive use the I2S stream isn't the best because you will have some delays between any action and it appearing on the pin (If you look in the issues you might find the calculations for this). For quite a few things it doesn't matter likes the mosfets, but for communication or as a reaction to async events it's not great.

petervanderwalt, 2021-11-12:
Thank you for getting back to me! (: 

> Did you configure Marlin properly to have the first 16 bits as the stepper stream and the other 16 bits as the "pins"? If you have the default configuration IIRC all 32bits will be the steppers stream which won't allow you to use them as GPIOs.

Aha!  Did not know about that - I've not found much documentation for the I2S stuff in Marlin, can you point me to details on this?

> Out of the box no, but making the I2S code work over 64 bits instead of 32 should be fairly straightforward.
> 
> >
Ahh that explains my other issues then (:  Do you think you can point me or help me achieve that? 

And do you have a Paypal donation link by any chance?

simon-jouet, 2021-11-15:
> Aha! Did not know about that - I've not found much documentation for the I2S stuff in Marlin, can you point me to details on this?

You need to have I2S_STEPPER_SPLIT_STREAM defined and it will split the channels into dynamic (the stepper data) and constant (the GPIOs). https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/ESP32/i2s.cpp#L335

>  I've not found much documentation for the I2S stuff in Marlin

Yeah I've never really written documentations unfortunately, it might be something that happens in the future.

> Ahh that explains my other issues then (: Do you think you can point me or help me achieve that?

I haven't tested yet (I can have a go if needed), but if your look at the ESP32 TRM you will see (page 311) that if Tx FIFO mode0 is set to 2 then it switches to a 32-bit dual channel data. At the moment it's a 16-bit dual channel data (32 bits), so changing to to mode 2 should switch to 64 bits.

So is looks like changing https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/ESP32/i2s.cpp#L266 from 0 to 2 will allow you to have 64 bits. If you enable the Stepper_split_Stream like above, then you will have 32 bits of stepper stream and 32 for GPIOs.

You could potentially decouple that and use all 64 bits for either but the latency is going to be higher (because in contant mode the next i2s clock will get that data, in stereo mode used for the stepper you will have the latency of the DMA buffers which will be significantly longer)

closed 2023-06-21 by petervanderwalt